getreuer/qmk-keymap #2: Overflow? issue

drashna, 2021-12-13:
This is more of a minor thing... but if the minimum length is 3 or lower, it causes issues. Odd output or hard locks on the board.  I don't think that fixing it is worth it, but adding an error or pragma message to warn about it may be a good idea. 

Eg, after the include for the data, something like this. 
```c
#    if AUTOCORRECTION_MIN_LENGTH < 4
#        error Minimum Length is too short and may cause overflows
#    endif
```

getreuer, 2021-12-14:
Thanks for pointing this out! I just pushed a commit with a compile-time check as you suggest.

I didn't manage to repro this yet. Could you share the dict file that you saw these problems with?

drashna, 2021-12-15:
Mostly, it was adding `gt: -> git` that did it.

closed 2022-03-14 by drashna