getreuer/qmk-keymap #4: Migrate autocorrect to QMK Core?

drashna, 2021-12-31:
Would you be opposed if I migrated the code into QMK core? 
It's a very useful feature, and I think that many more people would benefit from it.  Also, the python can be integrated into the qmk utility.

getreuer, 2022-01-01:
Awesome, migrating into QMK core would be great. My only concern is license. Would this migration be possible with the [current Apache license](https://github.com/getreuer/qmk-keymap/blob/main/LICENSE.txt)? The code is Google-owned, and they prefer to open source with Apache license. Let me know if that would work.

Other than that, I am fully in support! Thanks for suggesting this.

drashna, 2022-01-02:
QMK uses gpl2+ specifically, which is compatible with apache sub-licensing.  So it's definitely allowable. 

Also, with help, I have been able to migrate the database generation code into qmk-cli (as that is already python).

I've opened a PR for qmk firmware, and you can take a look at it over here: 
https://github.com/qmk/qmk_firmware/pull/15699

getreuer, 2022-01-02:
Fantastic! Really cool to see how you weave it into QMK. I'll make a review pass on your PR.

closed 2022-03-02 by drashna