getreuer/qmk-keymap #7: Achordion seems to break modifiers on combo keys

jdart, 2022-03-24:
I used Achordion today and it worked great for the most part but I wasn't able to use mod-tap modifiers with key codes I trigger with a combo. I try to avoid lateral movement with my index fingers, so I use combos for the inner columns of my colemak layout. 

For example I press `f`+`p` to produce a `b`, but when I have `LSFT_T(KC_S)` held, the combo doesn't work and `f`+`p` are output.

Here's a [link to my keymap](https://github.com/jdart/qmk_firmware/blob/master/keyboards/ferris/keymaps/dart-hmr/keymap.c#L60). Where you can see how I set things up.

getreuer, 2022-03-24:
Thanks for the clear report and for sharing your keymap. Much appreciated! I'll look into it.

getreuer, 2022-03-25:
Sorry, I don't see a way to get Achordion to work properly together with Combos. Combos are handled before the point where userspace code can intercept them, and Combos don't expose APIs that help with this. You could still use Combos and Achordion in the same keymap if combos are pressed only while tap-hold keys aren't held, though that would be hard to do with your keymap, unfortunately. Thanks again for your report.

closed 2022-03-25 by getreuer

getreuer, 2022-03-25:
On second thought, it is possible to detect when an event comes from a Combo and avoid having Achordion act on it. I just pushed a commit that does that. I don't expect this fully solves interoperation with Combos, but in my testing it does seem to help. I hope this works for you too.

jdart, 2022-03-25:
Thanks for the update, it's working for all the cases I've tried so far 🔥

getreuer, 2022-03-25:
Fantastic! Glad to hear it =)