getreuer/qmk-keymap #12: Caps word shifts `KC_MINS` on layer to `_`

0x64746b, 2022-05-12:
Hi Pascal,

thanks for sharing your _really_ nice QMK features! I just tried the caps word feature and stumbled upon `KC_MINS` (which lives on a layer in my keymap) to be shifted to `_`. Is this the expected behavior? Am I holding it wrong? :)

Thanks again!
D.

0x64746b, 2022-05-12:
D'oh, I saw both the documentation _and_ that particular piece of code, but I somehow failed to parse the combined switch cases and the comments 🤦 
Personally, I _think_ I prefer to be able to still type all symbols while caps lock is on, but must admit that that might just be old thinking from when toggling caps lock was a nuisance and I tried to get away with typing operators between consecutive constants 😊
Thanks for the super quick reply!!

getreuer, 2022-05-12:
> Personally, I think I prefer to be able to still type all symbols while caps lock is on

I agree it's not  a clear choice, especially between `_` vs. `-` when writing code. I think `_` and `-` are both so useful that I have dedicated two separate buttons for them on the base layer.

0x64746b, 2022-05-12:
> I think _ and - are both so useful that I have dedicated two separate buttons for them on the base layer.

Uuh, nice! I'm not brave enough for that ;) For all I have customised over the years (Colemak, column stagger, the custom symbols layers that come with 3x5 boards,...) I have a strange attachment to _somewhat_ being able to occasionally switching to a notebook keyboard (which is a bit nonsensical - you won't catch me typing on qwerty, and even just sliding the numbers row down by one on my daily driver, even while preserving finger assignments, means lots of weird typos on a regular keyboard. Not to mention home row mods and thumb cluster). But what do you know 🤷‍♂️