getreuer/qmk-keymap #16: compiling hex files brings up undefined reference to caps_word_on

morphykuffour, 2022-07-13:
I am trying to compile your qmk-keymaps to a hex file in order to flash it to my dactyl manuform but I am an undefined reference to `caps_word_on` function.  The only changes I made to your repo was including your custom header for the caps word header file and define the `QK_BOOT` macro as `RESET` because I was getting implicit definition errors also. Please take at the images I have below. Is there something I am missing or messing up to get these errors and not be able to compile to the hex file. All your help would be greatly appreciated.

[image]

[image]

[image]

getreuer, 2022-07-13:
Thanks for the report! Appreciate the detail and screenshots.

I think there are two things going on.

* About `QK_BOOT`: QMK recently had a ["breaking changes" release](https://docs.qmk.fm/#/ChangeLog/20220528) in which they deprecated "`RESET`" as a keycode name in favor of a new name "`QK_BOOT`." To get this definition, please update your QMK set up with
  ``` 
  git checkout master
  git pull upstream master
  ```
  Or see QMK's [Git Best Practices page](https://docs.qmk.fm/#/newbs_git_using_your_master_branch?id=updating-your-master-branch) for detailed tips on working with git.

* About Caps Word: in that same release, QMK added Caps Word as a core feature. It works nearly the same as the userspace `feature/caps_word.h` implementation in this repo (I am author of both), so if you were using that before it is hopefully an easy switch. After updating, you can enable Caps Word just by adding in rules.mk:

  ```
  CAPS_WORD_ENABLE = yes
  ```

  Further details: [QMK Caps Word documentation](https://docs.qmk.fm/#/feature_caps_word)

  Alternatively, you can continue to use the userspace Caps Word in this repo instead. In that case, please [follow the instructions here](https://getreuer.info/posts/keyboards/caps-word/index.html#add-caps-word-to-your-keymap). Be sure to change rules.mk as:

  ```
  # Enable userspace Caps Word implementation.
  SRC += features/caps_word.c
  
  # Disable QMK core's Caps Word implementation.
  CAPS_WORD_ENABLE = no  
  ```

morphykuffour, 2022-07-14:
Great I finally got it to work! I appreciate all your had work.

closed 2022-07-14 by morphykuffour

getreuer, 2022-07-14:
Excellent! Glad to hear it's all working.