getreuer/qmk-keymap #36: Overriding Space key in code doesn't trigger word separation

GraffJosh, 2023-06-08:
Hey there! In my code I use custom keys on my spacebars, and find that using "tapcode(kc_spc)" to send the space to the PC doesn't send it to autocorrect as well. Is there a workaround for this (I'm about to try register_code instead)? 

Thanks!

getreuer, 2023-06-11:
Hi @GraffJosh ! Yes, that's me. Thanks for checking out Autocorrect.

The `tap_code` and `register_code` APIs both directly send keycodes to the host without QMK features processing them, for better or worse. So yes, unfortunately for Autocorrect, it won't know about `tap_code` or `register_code` calls, and that could make a difference in whether what is being written is a typo.

The following should work, though: instead of calling `tap_code(KC_SPC)` on its own, do

```c
// Copyright 2023 Google LLC.
// SPDX-License-Identifier: Apache-2.0
if (process_autocorrect(
    KC_SPC, &(keyrecord_t){.event = MAKE_KEYEVENT(0, 0, true)})) {
  tap_code(KC_SPC);
}
```

Explanation: This simulates what happens when `KC_SPC` is pressed as a regular key. It calls the Autocorrect feature's [process handler](https://github.com/qmk/qmk_firmware/blob/master/quantum/process_keycode/process_autocorrect.h#L10) with the `KC_SPC` keycode that we're about to tap (and a dummy `keyrecord_t`) so that it is made aware of the key. The return value from a process handler is whether subsequent processing for that key event should continue. Autocorrect will usually return true, but if it returns false, this means Autocorrect wants to block the `KC_SPC` key as part of making some typo correction, so don't tap `KC_SPC` in that case.

GraffJosh, 2023-06-11:
Oh yeah, that makes a huge difference, thanks for your help; everything is working perfectly now.

closed 2023-06-11 by GraffJosh