getreuer/qmk-keymap #42: [Bug] Word gets inverted when sentence case is active and manual SHIFT

dd871, 2023-08-29:
Love the sentence case feature and recently added it to my keyboard.

The thing that keeps tripping me several times each day though is that whenever I forget that sentence case exists - and I hence manually SHIFT at the beginning of a new sentence - it inverts the complete word.

E.g. I get "tHE" instead of "The".

I have a suspicion that this is some weird behaviour when QMK sees a double-shift, but I cant figure out how I could disable this and are especially confused why it inverts the whole word, rather than only the first character.

So hence might not really be a bug, but was hoping for an idea how I could avoid this.

getreuer, 2023-08-29:
@dd871 thanks for checking out Sentence Case and reporting this.

It's peculiar that shifting is getting inverted. At a low level, QMK does bitwise-or of the different kinds of mods (one-shot mods, weak mods, ...) so that the usual behavior is shift on top of another shift produces shift.

Does the shifting return to normal at the end of the word, or does it remain stuck in this flipped state? I'm interested in what is making it "unstuck", since that's a good clue about the root cause.

Do you happen to use Caps Word? Perhaps with the "CAPS_WORD_INVERT_ON_SHIFT" option?

Do you happen to use Key Overrides?

What kind of shift keys are you using? the vanilla `KC_LSFT` / `KC_RSFT`, or is it something else like mod-tap, one-shot, or Space Cadet shifts?

Does your computer run key mapping software like AutoHotkey or Karabiner that may have an effect on Shift?

Thanks again for the report!

dd871, 2023-08-30:
Thank you very much for the quick reply and possible reasons why this might happen. This gave me the right hints to identify the root cause.

I had set BOTH_SHIFTS_TURNS_ON_CAPS_WORD in my config. This obviously then led to CAPS LOCK being activated for one word whenever I pressed RIGHT SHIFT in addition to a trigger of sentence case.

After removing this from my config, it all works well now.

Thanks again!

closed 2023-08-30 by dd871