getreuer/qmk-keymap #43: Custom shift keys for left shift only?

humanplayer2, 2023-09-18:
Thank you for Custom shift keys! I followed your [post](https://getreuer.info/posts/keyboards/custom-shift-keys/index.html), and it's great! It really helps this QMK noob out!

I have duplicated the functions so I can set also custom `LGUI` and `LCTL` keys, hoping I could retain a "pure" control key through `RCTL`, but that key is also affected (as is `RSFT`).

Is it possible to tweak it so it does not affect `MOD_RSFT` and in general differentiate between the left and right versions of mods?

humanplayer2, 2023-09-18:
Yes, you can!

Everywhere in `custom_shift_keys.c`, replace `MOD_MASK_SHIFT` with `MOD_BIT(KC_LEFT_SHIFT)`.

I found the answer in one of [your other blog posts](https://getreuer.info/posts/keyboards/triggers/index.html#when-mod-combo-is-held).

closed 2023-09-18 by humanplayer2

getreuer, 2023-09-19:
Yes, exactly! Thanks for using Custom shift keys and posting this idea and solution.