getreuer/qmk-keymap #51: [Bug] Achordion disables RETRO_TAPPING

wolfwood, 2023-12-14:
## Describe the bug

Single mod-tap key holds don't produce the tap key, with `RETRO_TAPPING`, when Achordion is enabled.

I'm using a new keyboard, which I am much slower to type on. I have previously used Homerow mods quite happily on both QMK and ZMK.

I noticed after enabling homerow mods, eg `LALT_T(KC_A)`, that my normal tapping strokes on the home row weren't reliably producing keystrokes anymore. I read up on QMK's tap-hold implementation and learned that by default mod-tap holds do not register if no second key is pressed. I assumed I was simply holding too long and that was swallowing my keystrokes.  `RETRO_TAPPING` to the rescue. This did what it said and transmitted my tap keys when holds were released, provided no other key was pressed.

However, then I enabled Achordion, and the missing keystrokes on single mod-tap holds returned.

## Information

**Do the keys involved use any of the following features?**
 - [X] Achordion (from this repo)
 - [ ] Auto Shift
 - [ ] Combos
 - [ ] Key Overrides
 - [X] Mod-tap or Layer-tap keys
 - [ ] Other custom userspace code:

akaralar, 2024-01-05:
I have the same problem. I'm using Retro Tapping for a single key and the key does not register as a tap unless released after Achordion timeout. I saw the blog post says it isn't supported but it also says it should be possible to implement. I checked the code and changing 
```c
if (achordion_state == STATE_HOLDING) {
```
to 
```c
if (achordion_state == STATE_HOLDING || IS_RETRO(tap_hold_keycode)) {
```
on line 135 seems to have fixed it for me. I'm not sure if I'm considering every edge case though, if @getreuer would be so kind to provide some feedback whether it's a viable workaround would be great!

akaralar, 2024-01-06:
After around a day of usage, I kinda feel that what I've done sometimes cancels the correct behaviour of Achordion 🤔

akaralar, 2024-01-07:
@wolfwood just so you know I opened a PR for this that fixed retro tap keys for me, let me know if it also fixes it for you

wolfwood, 2024-01-08:
@akaralar I saw your earlier message about cancelling correct achordion behavior. did you change something to address that?

akaralar, 2024-01-08:
@wolfwood it turned out to be unrelated to what I did, as when I took it back the issue was still happening, hence I deleted that message