getreuer/qmk-keymap #62: how is Layer Lock different than TO() ?

mmikeww, 2024-04-18:
i'm reading the layer lock blog post, but when Layer2 is being held down, why couldn't the layer lock key simply be a built in `TO()` action?

getreuer, 2024-04-19:
Sure thing, happy to clarify. The use case that Layer Lock has in mind is a layer that:

1. You usually want to activate "momentarily" by holding a `MO` layer switch (or other switch that is held like an `LT`, `OSL`, or `LM`),

2. Yet sometimes you want to stay in that layer without having to hold a layer switch continuously.

Such situations come up especially on navigation layers and numpad layers on smaller keyboards. A user might want both #1 and #2 with a navigation layer for instance, between brief or extended uses or the arrow keys, or on a numpad layer for typing a single digit vs. a typing something long like a date or telephone number.

If only #2 were needed, then certainly, that could be done with simply a `TO` (or `TG`) switch.

To do both #1 and #2, it's possible to have both an `MO` and a `TO` on the base layer. This has downsides, however:

* This spends an additional base layer key, which is a real challenge on smaller keyboards.
* It's arguably complicating that the same layer is accessed in two different ways by two different keys.
* It is fairly easy to make a mistake when setting up a keymap with `TO` and get stuck in a situation where it is impossible to switch layers without rebooting the keyboard.

To be clear, I'm not saying `TO` is "wrong" to use. If it works for you, then use it! FWIW, I use `TO` pretty heavily in my own keymap. But yes, there are reasons why users might not be entirely happy with `TO` and want something else.

Layer Lock is an alternative without the downsides listed above: you can use one `MO` (or `LT`, etc.) on the base layer and place the Layer Lock key on the destination layer. The layer stays locked until the Layer Lock key or `MO` is pressed again. By design, the lock is easily "broken" to avoid layers from getting stuck.

mmikeww, 2024-04-20:
No no, i'm saying, we are on base layer 0. We then go to layer 1 by holding the key to `MO(1). And then while on layer 1, we have another key or same key to use `TO(1)` to stay on layer 1. Is that not possible?

getreuer, 2024-04-20:
Aha! Well, that's different. There is an elegance to what you describe and it could have been a nice way for `MO` and `TO` to work together, but no, I'm afraid that QMK doesn't work that way.

QMK's layer switches work in terms of setting or clearing bits in the current layer state:

* `TO(k)` sets the `k`th bit and clears all other bits on press ([here](https://github.com/qmk/qmk_firmware/blob/13434fc066ef0b213587f773cc05f1c5b7d31113/quantum/action.c#L602) in QMK core as an "ON_PRESS", "OP_BIT_SET" action).

* `MO(k)` sets the `k`th bit on press, and clears the `k`th bit on release ([here](https://github.com/qmk/qmk_firmware/blob/13434fc066ef0b213587f773cc05f1c5b7d31113/quantum/action.c#L635) as an "OP_ON_OFF" action). 

There is some further description about this in the [Keymap and Layers documentation](https://docs.qmk.fm/#/keymap?id=keymap-and-layers). It's all pretty low level and inherited from the older TMK firmware that QMK is based on.

Using `MO` and `TO` as you describe behaves in the following way:

1. `MO(1)` is pressed. The bit for layer 1 is set, activating the layer.
2. `TO(1) is pressed. The bit for layer 1 is set and clears all other bits. If other layers were on, they are now off. Layer 1 doesn't change, since its bit was already on.
3. `TO(1)` is released. This is a no-op.
4. `MO(1)` is released. The bit for layer 1 is cleared, **deactivating the layer** (!). All bits are now clear, and only layer 0 is active, rather than staying on layer 1.

To be clear, this is working as intended, not a QMK bug.

mmikeww, 2024-04-20:
ahhh thank you very much for the technical explanation. exactly what i was looking for. much appreciated

closed 2024-04-20 by mmikeww