getreuer/qmk-keymap #68: MODS_TO_NEUTRALIZE ignored by achordion_eager_mod()

Ceffa93, 2024-07-14:
My wish:
* I would like to eagerly activate LALT so I can use Alt+Click in Visual Studio code without delay.
* I would like to disable LALT release that activates special action in Chrome browser.

What I have tried:
* I set LALT in achordion_eager_mod() as eagerly activated
* I set LALT as MODS_TO_NEUTRALIZE in QMK

The problem:
LALT release action is invoked and not neutralized in either of these cases:
* After QMK timer elapses, and LALT is released
* Automatically after Achordion timer elapses

My quick solution:
Modify achordion to take neutralization settings into account:
```c
static void clear_eager_mods(void) {
  neutralize_flashing_modifiers(eager_mods); // edit
  unregister_mods(eager_mods);
  eager_mods = 0;
}
```

It works quite well but I did not look into all corner cases, I think it should be thought through properly and integrated.

getreuer, 2024-08-09:
@Ceffa93, I pushed commit https://github.com/getreuer/qmk-keymap/commit/744119420424f567bc937075fadd9772867d472e to revise how eager mods are handled:

* When an eager mod MT is settled as held, Achordion continues to hold the mod. The hold event is no longer plumbed to subsequent handlers.

* When an eager mod MT is settled as tapped, then Achordion clears the mod, now using flashing mods neutralization as suggested you suggested. A tap event is then plumbed to subsequent handlers.

I believe these changes may fix the problem you observed, but please let me know if not. Thank you for your help in investigating this issue!

MichaelOstermann, 2024-08-11:
Hi @getreuer!

I can observe a regression after 7441194: After applying an eager mod, any non-eager mods seem to no longer function

This is my config, I don't believe I'm using anything else that could influence this:

```c
bool achordion_eager_mod(uint8_t mod) {
    switch (mod) {
        case MOD_LSFT:
        case MOD_LCTL:
        case MOD_LALT:
        case MOD_LGUI:
            return true;

        default:
            return false;
    }
}

bool achordion_chord(uint16_t tap_hold_keycode,
                     keyrecord_t* tap_hold_record,
                     uint16_t other_keycode,
                     keyrecord_t* other_record) {
    return true;
}
```

As an example, these are the events I receive when pressing <kbd>LSFT</kbd>+<kbd>p</kbd>, followed by <kbd>RSFT</kbd>+<kbd>a</kbd>:

[image]

And here the same keypresses when reverting to 15146c3:

[image]

getreuer, 2024-08-11:
@MichaelOstermann thank you for the report! Apologies for this. I am able to repro what you observed. It appears the root cause is that the `eager_mods` variable is not always cleared appropriately on release. If a non-eager tap hold key is then held, Achordion fails to plumb the hold event. I'll push a fix for this soon.

getreuer, 2024-08-11:
@Ceffa93, @MichaelOstermann, I just pushed a fix https://github.com/getreuer/qmk-keymap/commit/37586e3de2de15b915855ecc866b46cbfbd63d23 for this hold event bug.

Additionally in looking at this, I realized Retro Tapping was bugged as well for eager mod MT keys. So I fixed that too.

@MichaelOstermann thanks again for catching that regression!

closed 2024-08-11 by getreuer