getreuer/qmk-keymap #71: [Bug] compilation errors after defining ACHORDION_STREAK

proostas, 2024-08-11:
## Describe the bug

ACHORDION_STREAK leads to the following compilation errors:

```
Compiling: users/proostas/features/achordion.c                                                     users/proostas/features/achordion.c: In function 'settle_as_tap':
users/proostas/features/achordion.c:122:23: error: 'keycode' undeclared (first use in this function)
  122 |   update_streak_timer(keycode, record);
      |                       ^~~~~~~
users/proostas/features/achordion.c:122:23: note: each undeclared identifier is reported only once for each function it appears in
users/proostas/features/achordion.c:122:32: error: 'record' undeclared (first use in this function)
  122 |   update_streak_timer(keycode, record);
      |                                ^~~~~~
users/proostas/features/achordion.c:123:7: error: 'is_streak' undeclared (first use in this function)
  123 |   if (is_streak && is_key_event && is_tap_hold && record->tap.count == 0) {
      |       ^~~~~~~~~
users/proostas/features/achordion.c:123:20: error: 'is_key_event' undeclared (first use in this function)
  123 |   if (is_streak && is_key_event && is_tap_hold && record->tap.count == 0) {
      |                    ^~~~~~~~~~~~
users/proostas/features/achordion.c:123:36: error: 'is_tap_hold' undeclared (first use in this function); did you mean 'is_tap_record'?
  123 |   if (is_streak && is_key_event && is_tap_hold && record->tap.count == 0) {
      |                                    ^~~~~~~~~~~
      |                                    is_tap_record
users/proostas/features/achordion.c:133:12: error: 'return' with a value, in function returning void [-Werror=return-type]
  133 |     return false;
      |            ^~~~~
users/proostas/features/achordion.c:97:13: note: declared here
   97 | static void settle_as_tap(void) {
      |             ^~~~~~~~~~~~~

```

I was setting Achordion according to [this great article](https://getreuer.info/posts/keyboards/achordion/index.html) and everything went smoothly until I defined ACHORDION_STREAK and this function:

```
uint16_t achordion_streak_chord_timeout(uint16_t tap_hold_keycode, uint16_t next_keycode) {
    // disable for layer-tap keys
    if (IS_QK_LAYER_TAP(tap_hold_keycode)) {
        return 0;
    }

    uint8_t mod = mod_config(QK_MOD_TAP_GET_MODS(tap_hold_keycode));
    // disable for shifts
    if (mod & (MOD_LSFT|MOD_RSFT)) {
        return 0;
    }

    return 150;
}
```

Is it a bug or I did something wrong?

## Information

**Do the keys involved use any of the following features?**
 - [x] Achordion (from this repo)
 - [ ] Auto Shift
 - [ ] Combos
 - [ ] Key Overrides
 - [ ] Mod-tap or Layer-tap keys
 - [ ] Other custom userspace code:

gazpachoking, 2024-08-12:
I think the logic got a bit re-arranged in the last commit, and the streak logic got separated from some variables it needs. Needs a bit of a fix to get the streak logic working again.

getreuer, 2024-08-12:
@proostas thanks for the report! Sorry about the churn.

> I think the logic got a bit re-arranged in the last commit, and the streak logic got separated from some variables it needs. Needs a bit of a fix to get the streak logic working again.

Yes, that's exactly it. I'm working on a fix now.

closed 2024-08-12 by getreuer