getreuer/qmk-keymap #74: [Bug] Shift does not end caps word when achordion is in use

gazpachoking, 2024-11-08:
## Describe the bug
When I'm in caps word mode, and I press my mod-tap shift, or type a shifted character, it doesn't end caps word mode when achordion is enabled. I am not 100% sure this isn't something to do with my own customization, but I've been trying to rule it out and haven't so far. If anyone can confirm this works fine for them I'll keep digging in to my own code.

e.g. I am trying to type `CAPSWord`
I hit \<capsword\>caps\<shift\>w\<shiftup\>ord
I end up with just `CAPSWORD`

## Information

**Do the keys involved use any of the following features?**
 - [x] Achordion (from this repo)
 - [ ] Auto Shift
 - [x] Combos
 - [x] Key Overrides
 - [x] Mod-tap or Layer-tap keys
 - [ ] Other custom userspace code:

gazpachoking, 2024-11-11:
Aha, seems this is only when using shift that has eager mods applied. I'll try to see what the fix is. Looks like we already have a workaround for caps word when invert on shift is enabled, maybe we need something similar for when it isn't enabled.

gazpachoking, 2024-11-11:
I just expanded the exclusion to always disable eager mods during caps word and everything works like I expect now.
```diff
- #if defined(CAPS_WORD_ENABLE) && defined(CAPS_WORD_INVERT_ON_SHIFT)
+ #if defined(CAPS_WORD_ENABLE)
```

getreuer, 2024-11-29:
Thank you for catching this and for suggesting the fix! I'll push a commit in a moment to do that.

closed 2024-11-29 by getreuer