getreuer/qmk-keymap #78: [Bug] Custom shift keys not working as expected

fasterius, 2025-01-14:
Thanks a lot for your amazing code and writing on all things typing-related! I've learnt a lot. I got my Voyager some months ago, while I've Oryx (ZSA's custom software) has been working great, I've wanted to add some more QMK-specific things such as custom shift keys, so here I am. My original `keymap.c` comes from the Oryx source, so I've only done some minor formatting changes and commenting out tap-dance-related things that won't be needed if the functionality desired here starts working. I'm totally new to QMK, so I'm sorry if the issue is something simple I've missed or done wrong.

## Describe the bug

Custom shift key does not work as desired: a mod-tap `:` key where `;` is by default, which produces `;` when shifted and `R-Ctrl` when held. Following the instructions at https://getreuer.info/posts/keyboards/custom-shift-keys/index.html does get me the MT correctly, but the `;` key still outputs `;`, while `shift + ;` yields `;` (so it even removed the default shifted version, which should give `:`).

I have a fork with the relevant code here: https://github.com/fasterius/qmk-firmware-zsa/tree/custom-shift-keys/keyboards/zsa/voyager/keymaps/fasterius. The relevant code looks like this:

```
##### rules.mk
CAPS_WORD_ENABLE = yes
COMBO_ENABLE = yes
COMMAND_ENABLE = no
CONSOLE_ENABLE = no
KEY_OVERRIDE_ENABLE = no
MOUSEKEY_ENABLE = no
ORYX_ENABLE = yes
RGB_MATRIX_CUSTOM_KB = yes
SPACE_CADET_ENABLE = no
# TAP_DANCE_ENABLE = no
SRC += features/custom_shift_keys.c
```

```
##### keymap.c
(...)
#define RCTL_COLN MT(MOD_RCTL, KC_COLN) // Used in the layout
(...)
#include "features/custom_shift_keys.h"
const custom_shift_key_t custom_shift_keys[] = {
  {MT(MOD_RCTL, KC_COLN), KC_SCLN}, // Shift : is ;
};
uint8_t NUM_CUSTOM_SHIFT_KEYS =
    sizeof(custom_shift_keys) / sizeof(custom_shift_key_t);
(...)
  if (!process_custom_shift_keys(keycode, record)) {
    return false;
  }
```

The latest versions of the `custom_shift_key.[h/c]` where copied from this repository.

## Information

**Do the keys involved use any of the following features?**
 - [ ] Achordion (from this repo)
 - [ ] Auto Shift
 - [ ] Combos
 - [ ] Key Overrides
 - [X] Mod-tap or Layer-tap keys
 - [ ] Other custom userspace code:

I've never coded in C and I'm totally new to QMK and everything related, so while I feel that I've done quite a bit of troubleshooting and trying to solve the issue myself I can't rule out that it's just a "user error" and that I've done something stupid. Sorry if that's the case!

gazpachoking, 2025-01-14:
I was about to tell you that I have a similar setup working fine, but I just looked at my keymap again and realized I have some kind of workaround in place. I don't remember exactly why I did it, but quite possibly it's the same issue you are having. My current workaround is this:
```c
// key definition
#define WIN_COLN MT(MOD_LGUI, KC_F24)
// custom shift
  {WIN_COLN, KC_SCLN}, // Shift : is ;
// in a switch statement in my process_record_user
    case WIN_COLN:
      if (record->event.pressed && record->tap.count == 1) {
        SEND_STRING(":");
		return false;
      }
      break;
```
https://github.com/gazpachoking/qmk-keymap/blob/mine/keyboards/zsa/voyager/keymaps/gazp/keymap.c#L39

getreuer, 2025-01-14:
@fasterius welcome to QMK! =)

If I got that correctly, you want to make a mod-tap `MT` key that acts as 

* right Ctrl when held
* colon `:` when tapped
* semicolon `;` when tapped with Shift

The trouble is unrelated to Custom Shift Keys, I think. Mod-tap (and similarly layer-tap) keys are limited to [basic keycodes](https://docs.qmk.fm/keycodes_basic) for the tapping keycode, unfortunately. Shifted symbols like `:` (= Shift + `;`) are not basic. Attempting a keycode like "`MT(MOD_RCTL, KC_COLN)`" does not work. In fact (if one digs up the definition of the `MT` macro), it works out such that `MT(MOD_RCTL, KC_COLN)` masks off the Shift modifier on `;`, and it ends up equivalent to keycode `MT(MOD_RCTL, KC_SCLN)`, which does `;` on tap.

It sounds like Custom Shift Keys then forced Shift + `MT(MOD_RCTL, KC_SCLN)` to produce `;`, which is doing its part as intended.

## Possible solution

It is not possible to write a tap-hold keycode for a non-basic tapping key. But a workaround is possible (["changing the tap function"](https://docs.qmk.fm/mod_tap#changing-tap-function)) by writing a small amount of code, like @gazpachoking said. There are some more examples [on this page](https://getreuer.info/posts/keyboards/faqs/index.html#mt-doesnt-work-with-this-keycode-qmk).

The special behavior with Shift might be implemented at the same time, instead of using Custom Shift Keys, if you find it simpler that way. See [macros that respond to mods](https://getreuer.info/posts/keyboards/macros/index.html#macros-that-respond-to-mods) for some examples of how to do that.

Maybe something like...

```c
// Copyright 2025 Google LLC.
// SPDX-License-Identifier: Apache-2.0

#define CTL_COL RCTL_T(KC_0)

// Use CTL_COL in your layout...

bool process_record_user(uint16_t keycode, keyrecord_t* record) {
  const uint8_t saved_mods = get_mods();
  const bool shifted = (saved_mods | get_oneshot_mods()) & MOD_MASK_SHIFT;

  switch (keycode) {
    // When held: right Ctrl.
    // When tapped: KC_COLN (:).
    // When tapped + Shift: KC_SCLN (;).
    case CTL_COL:
      if (record->tap.count) {  // On tap.
        if (record->event.pressed) {  // On press.
          // Make sure Shift is cleared.
          del_oneshot_mods(MOD_MASK_SHIFT);
          unregister_mods(MOD_MASK_SHIFT);
          // Tap the desired key.
          tap_code16(shifted ? KC_SCLN : KC_COLN);
          // Restore the mods.
          set_mods(saved_mods);
        }
        return false;  // Skip default handling.
      }
      break;  // Continue default handling for Ctrl on hold.

    // Other macros...
  }
  return true;  // Continue default handling.
}
```

fasterius, 2025-01-15:
That's awesome, thanks a lot for the help! That was quite a lot less code than was is included in `custom_shift_keys`, but I'm guessing that this doesn't scale well compared to that for more than one keys?

Just so I'm understanding correctly: clearing Shift is so that whenever a `KC_COLN` is sent it is sent properly without the previous issues with it already being a "shifted" key?

getreuer, 2025-01-15:
> clearing Shift is so that whenever a KC_COLN is sent it is sent properly without the previous issues with it already being a "shifted" key?

The other way around: the issue is with the `;` being an unshifted keycode. Suppose we have a sequence like this, where we want to send `;`

1. The user holds the Shift key.
2. The user taps the `CTL_COL` key.

The gotcha is that when `tap_code16(kc)` is called, it sends a tap of `kc` plus any mods that were already active. If we didn't clear the Shift mod first, the `;` would be shifted to produce `:`.

> That was quite a lot less code than was is included in `custom_shift_keys`, but I'm guessing that this doesn't scale well compared to that for more than one keys?

Right, once you know the pattern to it, it's straightforward enough to code directly yourself. `custom_shift_keys` does a few small things beyond this and wraps this behavior in an interface for ease of use, which adds up in lines of code. 

To scale to multiple keys, for starters, it's fine IMO to duplicate the handler. This way you can flexibly add further customizations for each key independently. If it turns out a good number of keys follow the same pattern, it could be useful to parameterize and extract a helper function. Perhaps like

```c
// Copyright 2025 Google LLC.
// SPDX-License-Identifier: Apache-2.0

// Handle a mod-tap or layer-tap key, overriding the tap function while leaving the
// hold function as is. When tapped, the key sends `unshifted_tap_keycode`. When tapped
// with Shift, the key sends `shifted_tap_keycode`. Notably, these keycodes can be
// anything that tap_code16() can send, which includes shifted basic keycodes.
static bool handle_tap_hold_key_custom_shift(
    keyrecord_t* record, uint16_t unshifted_tap_keycode, uint16_t shifted_tap_keycode) {
  // Continue default handling on hold.
  if (record->tap.count == 0) { return true; } 

  const uint8_t saved_mods = get_mods();
  const bool shifted = (saved_mods | get_oneshot_mods()) & MOD_MASK_SHIFT;

  if (record->event.pressed) {  // On press.
    // Make sure Shift is cleared.
    del_oneshot_mods(MOD_MASK_SHIFT);
    unregister_mods(MOD_MASK_SHIFT);
    // Tap the desired key.
    tap_code16(shifted ? shifted_tap_keycode : unshifted_tap_keycode);
    // Restore the mods.
    set_mods(saved_mods);
  }

  return false;  // Skip default handling.
}

bool process_record_user(uint16_t keycode, keyrecord_t* record) {
  switch (keycode) {
    case CTL_COL:
      return handle_tap_hold_key_custom_shift(record, KC_COLN, KC_SCLN);
    // ...
```

fasterius, 2025-01-16:
Thanks a lot for this, I appreciate it!

closed 2025-01-20 by getreuer