getreuer/qmk-keymap #81: [Bug] get_chordal_hold_default not found

jsulopzs, 2025-02-18:
I'm compiling the repo for the voyager:

```bash
qmk flash -kb zsa/voyager -km getreuer
```

And I get this error:

```bash
Ψ Compiling keymap with gmake -r -R -f builddefs/build_keyboard.mk -s flash KEYBOARD=zsa/voyager KEYMAP=getreuer KEYBOARD_FILESAFE=zsa_voyager TARGET=zsa_voyager_getreuer VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk" QMK_USERSPACE=/Users/sota/qmk/qmk-keymap-getreuer MAIN_KEYMAP_PATH_1=/Users/sota/qmk/qmk-keymap-getreuer/keyboards/zsa/voyager/keymaps/getreuer MAIN_KEYMAP_PATH_2=/Users/sota/qmk/qmk-keymap-getreuer/keyboards/zsa/voyager/keymaps/getreuer MAIN_KEYMAP_PATH_3=/Users/sota/qmk/qmk-keymap-getreuer/keyboards/zsa/voyager/keymaps/getreuer MAIN_KEYMAP_PATH_4=/Users/sota/qmk/qmk-keymap-getreuer/keyboards/zsa/voyager/keymaps/getreuer MAIN_KEYMAP_PATH_5=/Users/sota/qmk/qmk-keymap-getreuer/keyboards/zsa/voyager/keymaps/getreuer

Generating: .build/obj_zsa_voyager_getreuer/src/info_deps.d                                         [OK]
arm-none-eabi-gcc (Homebrew ARM GCC 8.5.0_2) 8.5.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_zsa_voyager_getreuer/src/info_config.h                                       [OK]
Generating: .build/obj_zsa_voyager_getreuer/src/default_keyboard.c                                  [OK]
Generating: .build/obj_zsa_voyager_getreuer/src/default_keyboard.h                                  [OK]
Compiling: keyboards/zsa/voyager/matrix.c                                                           [OK]
Compiling: drivers/gpio/mcp23018.c                                                                  [OK]
Compiling: /Users/sota/qmk/qmk-keymap-getreuer/features/custom_shift_keys.c                         [OK]
Compiling: /Users/sota/qmk/qmk-keymap-getreuer/features/keycode_string.c                            [OK]
Compiling: /Users/sota/qmk/qmk-keymap-getreuer/features/orbital_mouse.c                             [OK]
Compiling: /Users/sota/qmk/qmk-keymap-getreuer/features/sentence_case.c                             [OK]
Compiling: /Users/sota/qmk/qmk-keymap-getreuer/features/select_word.c                               [OK]
Compiling: keyboards/zsa/voyager/voyager.c                                                          [OK]
Compiling: .build/obj_zsa_voyager_getreuer/src/default_keyboard.c                                   [OK]
Compiling: quantum/keymap_introspection.c                                                          In file included from /Users/sota/qmk/qmk-keymap-getreuer/keyboards/zsa/voyager/keymaps/getreuer/keymap.c:27,
                 from quantum/keymap_introspection.c:5:
/Users/sota/qmk/qmk-keymap-getreuer/getreuer.c: In function 'get_chordal_hold':
/Users/sota/qmk/qmk-keymap-getreuer/getreuer.c:332:10: error: implicit declaration of function 'get_chordal_hold_default'; did you mean 'get_chordal_hold'? [-Werror=implicit-function-declaration]
   return get_chordal_hold_default(tap_hold_record, other_record);
          ^~~~~~~~~~~~~~~~~~~~~~~~
          get_chordal_hold
In file included from /Users/sota/qmk/qmk-keymap-getreuer/keyboards/zsa/voyager/keymaps/getreuer/keymap.c:27,
                 from quantum/keymap_introspection.c:5:
/Users/sota/qmk/qmk-keymap-getreuer/getreuer.c: At top level:
/Users/sota/qmk/qmk-keymap-getreuer/getreuer.c:819:9: note: #pragma message: dlog_record: enabled
 #pragma message "dlog_record: enabled"
         ^~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
gmake: *** [builddefs/common_rules.mk:362: .build/obj_zsa_voyager_getreuer/quantum/keymap_introspection.o] Error 1
```

getreuer, 2025-02-18:
@jsulopzs good question! I need to document this more prominently: I recently developed [Chordal Hold](https://github.com/qmk/qmk_firmware/blob/develop/docs/tap_hold.md#chordal-hold) as something of a successor to Achordion as a core QMK feature and have started using it in my keymap. The build problem is because Chordal Hold hasn't released yet on QMK's master branch.

To build the keymap, do one of...

* Use QMK's develop branch and pull the latest (`git checkout develop && git pull`).
* Or wait until 2025-02-23, when QMK makes its next breaking changes release.
* Or you can disable Chordal Hold by removing `#define CHORDAL_HOLD` from config_getreuer.h.

jsulopzs, 2025-02-18:
Awesome, thanks @getreuer 

So far, I had placed a `return true` as per AI suggestion, which made the trick. But I'll wait until QMK next release for this case.

closed 2025-02-18 by jsulopzs